ykasidit/bluetooth_gnss #2: Add app to f-droid

jensMF, 2021-03-13:
Hi,

I really like your App and since the only open source alternative I found is outdated, I think it would be great to have your App on the f-droid store. (https://f-droid.org/)

(What is F-Droid?
F-Droid is an installable catalogue of FOSS (Free and Open Source Software) applications for the Android platform. The client makes it easy to browse, install, and keep track of updates on your device.)

The inclusion policy of f-droid (https://f-droid.org/en/docs/Inclusion_Policy/) states:
"For software to be FLOSS, the software in its entirety must be so - including all libraries and dependencies used. Additionally, it must be buildable with only FLOSS tools."

If this is the case for your App and you like the idea that the App is on the f-droid store you or I could submit a "Request For Packaging" or you could put together the relevant metadata and submit that via the F-Droid Data repository, which will drastically speed up the inclusion of the application. (https://f-droid.org/en/contribute/)

All in all there is no action required by you, if you support the inclusion to f-droid. I could submit the request for packaging, but of course I let you decide. You can also take care of it by yourself or decide to not have it on f-droid.

Thank you in advance.

ykasidit, 2021-04-07:
Hello jens,

Thanks for the suggestion! Thanks for your offer to help submit the request, totally agree as this project and its engine (https://github.com/ykasidit/libbluetooth_gnss) are all released under the GPL so it should be fine, please help do so anytime you find comfortable.

closed 2021-04-07 by ykasidit

jensMF, 2021-06-12:
I created the "Request for Packaging" issue: https://gitlab.com/fdroid/rfp/-/issues/1768