ykasidit/bluetooth_gnss #7: Provide informtion how to use Auto-reconnect with open source App

jensMF, 2021-06-27:
Could you provide information on how to use the Auto-reconnect mode with different apps?
Especially with different free/libre open source apps like Automation (https://server47.de/automation/) or Easer (https://me.ryey.icu/Easer/en/).

Thanks in advance.
@auricgoldfinger

pyrog, 2021-07-08:
What I observed is that the app automatically start and try to connect to the GNSS when the phone restart.

ykasidit, 2021-07-11:
Another developer/contributor, auricgoldfinger, implemented the auto start
on boot feature and he said he uses it with 'tasker' but I don't know how
to use it myself...

auricgoldfinger, 2021-07-11:
Yes, it should be relatively easy to use in Automate as well but I didn't have the chance to set up a flow. I'll look into it as soon as I can

jensMF, 2021-08-11:
I think this is not necessary anymore. It is solved with #8 I think.

closed 2021-08-11 by jensMF