ykasidit/bluetooth_gnss #11: Redesign main screen

pyrog, 2021-07-08:
Hide  "usage" informations or move them i.e. the About menu.

[image] [image]

In practice, especially when driving a car or a bike, this is more useful to see main informations:

- mount point (or station ID)
- baseline (distance between the rover and the base)
- Horizontal / vertical / 3D precision (2D RMS)
- fix type (fixed, float, dgps…)
- …

For example, see [Lefebure NTRIP client](https://play.google.com/store/apps/details?id=com.lefebure.ntripclient):
[image]

ykasidit, 2021-07-11:
Thanks for the suggestions!

ykasidit, 2021-09-11:
Moved main params part to top.

closed 2021-09-11 by ykasidit