ykasidit/bluetooth_gnss #9: Sort mountpoints by distance

pyrog, 2021-07-08:
During a trip this is difficult to find the **nearest base**, especially with a long mount list (e.g. [centipede](http://caster.centipede.fr:2101) with 160 bases)
It will be useful to have a sort option (by name, distance…) and to display the distance in the list.

[image]

ykasidit, 2021-07-11:
Thanks for the suggestion, will try sort it. (Not sure when though as my
main work is quite demanding now, badly affected by covid...)

pyrog, 2021-07-12:
Ok Kasidit, we understand 😃 

Mountpoints are queried by [MainActivity.java#L110](https://github.com/ykasidit/bluetooth_gnss/blob/7f500fe6f77836fabd12f5568a05be4ac5dc181a/android/app/src/main/java/com/clearevo/bluetooth_gnss/MainActivity.java#L110)
List is build and sorted by [settings.dart#L93](https://github.com/ykasidit/bluetooth_gnss/blob/076ba76094272b8b06bb614e5c1eae2a34279231/lib/settings.dart#L93)

ykasidit, 2021-07-24:
Ok trying something on this saturday morning haha - not released yet, there would be a checkbox like this:
[image]

So it would sort and show like this:
[image]

Will try update again when this is released.

ykasidit, 2021-07-24:
Pls try the new apk from https://github.com/ykasidit/bluetooth_gnss/releases/tag/1.0.23 > app-release.apk and see if it works for you or not.

ykasidit, 2021-09-11:
Please help test and confirm if the release made it work on your side or not.

ykasidit, 2021-09-15:
Closing as assumed solved.

closed 2021-09-15 by ykasidit

pyrog, 2021-12-18:
I discovered this new feature yesterday. It works fine.
Thank you very much 😎