ykasidit/bluetooth_gnss #12: f-droid integration

jensMF, 2021-08-11:
Hi,

over at f-droid they wrote there are two things missing to add this app to the f-droid store:

- [ ] Add Fastlane structures to the repository as described [here](https://gitlab.com/-/snippets/1895688).
- [ ] Tag releases of the app, to make updates detectable.

Copied from there:
https://gitlab.com/fdroid/rfp/-/issues/1768#note_620025932

It would be great to have this app on f-droid.

aeri, 2021-08-30:
This application is great, necessary and functional and also open source, it would be super interesting its inclusion in F-Droid :)

Astrinus, 2022-08-21:
Hi, as mentioned in https://gitlab.com/fdroid/rfp/-/issues/1768#note_1071240633 F-Droid won't include BT GNSS in its collection if the two non-free dependencies (Play Services Places Placereport and Google Mobile Services) won't be removed or conditionally substituted with free open source alternatives.
Are they essential for BT GNSS to work, or they can be omitted for a special F-Droid flavor of the app?

Astrinus, 2022-09-04:
Ping.

ykasidit, 2022-10-29:
hello,

ok just removed them (there were required to get the last position in sort nearest ntrip feature - now this is changed to let user enter a ref lat lon instead if desired) - please hlpe try this new apk: https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.41

IzzySoft, 2023-06-06:
@ykasidit looks like this time you've missed the ping :smile: So I simply send another one. For background, there's an [RFP](https://gitlab.com/fdroid/rfp/-/issues/1768) open to include your app with F-Droid waiting for your response to be processed.

gdt, 2025-08-12:
Given the fixes for izzysoft, is this now feasible?