ykasidit/bluetooth_gnss #13: Log directory is not created on Android 11

gmatefi2, 2021-09-02:
When running Bluetooth GNSS on Android 11, the "bluetooth_gnss_logs' directory is not created and log files are not accessible. It is ok on Android 9.

 I suspect that the problem is related to "scoped storage" restrictions in Android 11. As in the Android SDK doc, from Android 11, apps are no longer permitted to create their own app-specific directories in the external storage but the system provides app-specific directories which are accessible with the getExternalFilesDirs() calls.

ykasidit, 2021-09-03:
Thanks! will try to look into it.

ykasidit, 2021-09-11:
Just coded this android 11 support - it should pop-up to ask you the target folder now on connection (android documentfile handling) - please help test and see if it works on your side:
https://github.com/ykasidit/bluetooth_gnss/releases/tag/1.0.25

gmatefi2, 2021-09-14:
Tested on an Android 11 device (Samsung A52). Upon selection of the target folder, a system window is popped up which asks for access permissions to that folder. When it is confirmed, the logs are written normally. Thanks for the quick fix!

ykasidit, 2021-09-15:
Great! Thanks for the testing!

closed 2021-09-15 by ykasidit