ykasidit/bluetooth_gnss #14: Listing nearest NTRIP streams first ?

HJarausch, 2021-10-02:
I have checked the _List nearest streams first_ checkbox.
But the first stream shown several thousands of kilometers apart.

( I have turned on GPS before s.t. my location is known)
What am I missing?

Thanks,
Helmut

grinapo, 2021-10-30:
How often do you release on GooglePlay?

ykasidit, 2021-12-11:
fyi, the fix is released on Google Play already now.

Mikixie, 2021-12-28:
tried it and still with the problem,maybe I use it in China ?
[image]
[image]

ykasidit, 2022-01-08:
Hello, this sorting feature was confirmed working already now but it will work only if your caster sends location in its stream list (not 0,0). In your screenshot seems like the issue is with reading the stream. I might need the login and pass for the ntrip caster to test it - pls share via email to ykasidit@gmail.com if you want me to check further.

ykasidit, 2022-02-04:
assume resolved

closed 2022-02-04 by ykasidit