ykasidit/bluetooth_gnss #15: [wishlist] play soundfile when fix quality changes

grinapo, 2021-10-30:
I would be neat to have a feature where I could assign (existing) sound files to playwhen fix changes. 

I imagine it to make an "frtk" sound when changing from dgps or from rtk, etc. So I could be alerted when precision is getting down without constantly checking the screen.

Similarly it would be useful to play some alert sound when network stream is broken  or reconnected successfully. 

Probably for you it'd be easiest to let the user assign the sounds, and do nothing if nothing's assigned.

Also probably there should be some cooldown time between alerts (say, 15 seconds, preferably a user setting).

Yogurt4, 2021-10-31:
Totally agree.

Use case: When I'm hiking on a difficult terrain or when climbing, I don't want to hold my phone in my hands to check the measurement quality. However, I need precise measurements as it's not likely I can/will go back easily again. But I can wait some time if I know the precision has dropped.

GSM signal is also an important aspect in the mountains. Without mobile network there is no RTK. So a broken connection invalidates the measurements.

jjojala, 2022-04-26:
Agree. Using this together with OpenOrienteering Mapper (OOM) for making maps. OOM uses (and for good reason) all the space on the screen. If I lose DGPS or connection to GNSS is lost (happens rarely with my setup) I quickly see that on the screen as the position starts to shake, but would be beneficial to get audio signal early.