ykasidit/bluetooth_gnss #18: Fails to parse RMC message

GeoSpark, 2021-12-17:
I'm not sure whether the fault is in bluetooth_gnss, libbluetooth_gnss, or marine-api, but my GNSS device outputs a 4.1 RMC which has the nav status field at the end. See https://gpsd.gitlab.io/gpsd/NMEA.html#_rmc_recommended_minimum_navigation_information
The error I get is:
```12-17 12:37:31.141  3618 31261 D btgnss_nmea_p: calling m_cb callback with parsed params
12-17 12:37:31.141  3618 31261 D btgnss_service: service: on_updated_nmea_params() start
12-17 12:37:31.141  3618 31261 D btgnss_service: ellips_height_key valid
12-17 12:37:31.142  3618 31261 D btgnss_service: bluetooth_gnss_service on_updated_nmea_params exception: java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
12-17 12:37:31.142  3618 31261 D btgnss_service: 	at com.clearevo.libbluetooth_gnss_service.bluetooth_gnss_service.on_updated_nmea_params(bluetooth_gnss_service.java:1165)
12-17 12:37:31.142  3618 31261 D btgnss_service: 	at com.clearevo.libecodroidgnss_parse.gnss_sentence_parser.parse_nmea_string(gnss_sentence_parser.java:308)
12-17 12:37:31.142  3618 31261 D btgnss_service: 	at com.clearevo.libecodroidgnss_parse.gnss_sentence_parser.parse(gnss_sentence_parser.java:95)
12-17 12:37:31.142  3618 31261 D btgnss_service: 	at com.clearevo.libbluetooth_gnss_service.bluetooth_gnss_service.on_readline(bluetooth_gnss_service.java:796)
12-17 12:37:31.142  3618 31261 D btgnss_service: 	at com.clearevo.libecodroidbluetooth.inputstream_to_queue_reader_thread.run(inputstream_to_queue_reader_thread.java:140)
12-17 12:37:31.142  3618 31261 D btgnss_service: service: on_updated_nmea_params() act
12-17 12:37:31.142  3618 31261 D btgnss_mainactvty: mainactivity on_updated_nmea_params()
12-17 12:37:31.142  3618 31261 D btgnss_service: service: on_updated_nmea_params() done
12-17 12:37:31.142  3618  3618 D btgnss_mainactvty: mainactivity handler got params map
12-17 12:37:31.142  3618  3618 D btgnss_mainactvty: sending params map to m_events_sink start
12-17 12:37:31.144  3618  3618 D btgnss_mainactvty: sending params map to m_events_sink done
```

Which essentially says that Speed is null:
```speed = (double) params_map.get(talker+"_speed"); //Speed in knots (nautical miles per hour).```

A typical RMC message is:
```$GNRMC,125645.000,V,5023.3510649,N,01020.1174608,W,000.0,000.0,171221,,,N,V*07```

ykasidit, 2021-12-17:
Hello,

Thanks for the report.

Please help collect and share your nmea (enable logging in settings) so I
can run the test against it.

GeoSpark, 2021-12-17:
Thanks for looking in to this. Here's a couple of minutes of data
[2021-12-17_13-40-09_rx_log.txt](https://github.com/ykasidit/bluetooth_gnss/files/7735508/2021-12-17_13-40-09_rx_log.txt)

ykasidit, 2021-12-19:
Hello,

Found that it didn't have errors with parsing the full RMC lines in
your log, the errors were instead for lines having somehow erroneously
'mixed' lines (two $) like below, might be some data/wire issues in
your device?
$GNRMC,1$GPGGA,134010.000,5347.3507666,N,00224.1163052,W,1,22,0.4,121.827,M,50.800,M,,0000*4B
$GNGLL,5347.3507666,N,00224.1163052,W,13401$GPGGA,134011.000,5347.3507645,N,00224.1163212,W,1,22,0.4,121.798,M,50.800,M,,0000*46

I added some code to skip these now.
Regarding the error/exception:

luetooth_gnss_service on_updated_nmea_params exception:
java.lang.NullPointerException: Attempt to invoke virtual method
'double java.lang.Double.doubleValue()' on a null object reference

Does this always come or only comes a few times at start?
As there's this loop:
for (String talker : GGA_MESSAGE_TALKER_TRY_LIST) {
I'm guessing the exception is not from the main GN message, or might
be that it's at the start of run where some vals might not be valid
yet, I added the exception to include the 'talker' to clear this up,
pls install the new app and share the new exception/error again, if
it's not GN then it's fine as your mean talker is GN.

The changes are now in below release, kindly give it a try:
https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.34

GeoSpark, 2021-12-20:
Ah! I assumed it was the RMC because of the speed being null, but that's just a symptom rather than the cause. As for the merged strings, there's not much I can do about that; the GNSS device talks to a ESP32 on the same board, and I just pull that to my phone over BLE. I can message the devs to see if there's anything they can suggest, if not, I am at a bit of a loss.
Thanks for putting in that fix, I'll test it today and get back to you. As for the error, I think I got it every update, but I'll check and post some timings.

GeoSpark, 2021-12-20:
That doesn't seem to have fixed it, I'm still getting *No data* despite having a "NORMAL" fix and 15 satellites in use.

The error is still present, and it appears every update:
```12-20 10:48:50.217 22142 22986 D btgnss_nmea_p: calling m_cb callback with parsed params
12-20 10:48:50.217 22142 22986 D btgnss_service: service: on_updated_nmea_params() start
12-20 10:48:50.217 22142 22986 D btgnss_service: ellips_height_key valid
12-20 10:48:50.218 22142 22986 D btgnss_service: bluetooth_gnss_service on_updated_nmea_params talker: GP exception: java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
12-20 10:48:50.218 22142 22986 D btgnss_service: 	at com.clearevo.libbluetooth_gnss_service.bluetooth_gnss_service.on_updated_nmea_params(bluetooth_gnss_service.java:1165)
12-20 10:48:50.218 22142 22986 D btgnss_service: 	at com.clearevo.libecodroidgnss_parse.gnss_sentence_parser.parse_nmea_string(gnss_sentence_parser.java:316)
12-20 10:48:50.218 22142 22986 D btgnss_service: 	at com.clearevo.libecodroidgnss_parse.gnss_sentence_parser.parse(gnss_sentence_parser.java:95)
12-20 10:48:50.218 22142 22986 D btgnss_service: 	at com.clearevo.libbluetooth_gnss_service.bluetooth_gnss_service.on_readline(bluetooth_gnss_service.java:796)
12-20 10:48:50.218 22142 22986 D btgnss_service: 	at com.clearevo.libecodroidbluetooth.inputstream_to_queue_reader_thread.run(inputstream_to_queue_reader_thread.java:140)
12-20 10:48:50.218 22142 22986 D btgnss_service: service: on_updated_nmea_params() act
12-20 10:48:50.218 22142 22986 D btgnss_mainactvty: mainactivity on_updated_nmea_params()
12-20 10:48:50.218 22142 22986 D btgnss_service: service: on_updated_nmea_params() done
12-20 10:48:50.218 22142 22142 D btgnss_mainactvty: mainactivity handler got params map
12-20 10:48:50.218 22142 22142 D btgnss_mainactvty: sending params map to m_events_sink start
12-20 10:48:50.222 22142 22142 D btgnss_mainactvty: sending params map to m_events_sink done
```

ykasidit, 2021-12-20:
Ok, now it is showing the exception's talker is 'GP', I checked your log
and now I understand that it is because your device is giving 'GNRMC' but
it is not giving 'GNGGA' (so it failed to match 'GGA' to 'RMC' of the same
'talker').
(Your nmea log shows there are only GPGGA and GNRMC so they will not match).

Please try to enable GNGGA (through the device config like u-center if it
is u-blox) and it should match the GNRMC and it should work then sir.

GeoSpark, 2021-12-20:
That fixed it! Thanks @ykasidit. I had to trawl through the manual, but I have found the relevant setting.

If anyone else finds themselves here, the command to send programmatically is _Configure NMEA talker ID_ *(0x4B)* on page 37 of this document: https://navspark.mybigcommerce.com/content/AN0037.pdf
In the GUI, it's in the Binary menu, right at the bottom.

closed 2021-12-20 by GeoSpark