ykasidit/bluetooth_gnss #22: NTRIP problem

Yogurt4, 2022-01-10:
Hi,

I've managed to configure the NTRIP settings. Yesterday I had "List streams from above server" working. (Now it doesn't work with the settings unchanged, by the way. It just says "Failed to list mount-points list from the server specified.")

However, after connecting my device, a small error message kept on popping up with HTTP 401 Unauthorized error.
I've double-checked my credentials, even registered a new account at www.euref-ip.net and have tried that one.

Is there anything I can check - from a browser or within the app?

I'm trying version 1.0.33 on an Android 10 phone.

pyrog, 2022-01-31:
Did you checked with another NTRIP client ?

Does Euref has banned this NTRIP client ?

Yogurt4, 2022-01-31:
I can use Lefebure NTRIP Client on the same phone. That app needs caster setup in a different way but it works with 193.190.230.76:2101 (that's www.euref-ip.be).

A friend of mine uses Bluetooth GNSS with 157.90.249.44:80 (e.g., www.euref-ip.net).
I've tried to copy his settings (except my credentials) but no luck.

ykasidit, 2022-02-04:
Hello, I tried your ip it seems the header before sourcetable is a bit different, now this is handled, please try see if new app apk solves your issue or not:
https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.39

If not solved then i might need a working login/pass to the server for testing - pls share to ykasidit@gmail.com if you want to try further.

Yogurt4, 2022-03-02:
Hi. Sorry for the delay. Just today I've received v1.0.39 from Play Store. I'll start to check when I get home.

Yogurt4, 2022-03-03:
Wow. After starting v1.0.39, I haven't had to modify or try anything, it just works. Thank you.

closed 2022-03-03 by Yogurt4