ykasidit/bluetooth_gnss #24: Quick question, what nema/ubx messages are needed to have all the data available?

lmore377, 2022-02-02:
I've tried enabling a bunch of different messages on my ublox neo-7m in u-center but no matter what I enable it just says no data under ublox fix type/accuracy and all the height values.

ykasidit, 2022-02-04:
Hello, I think its UBX_POSITION... Some ublox devices dont send these on UART2 even after config, but sends on UART1 and USB though. Please give it a try.

Joku-Suomalainen, 2023-06-16:
Had this problem myself. For me the fix was to enable $PUBX00 messages to uart2. 
u-center -> config -> messages -> F1-00 PUBX00 

> I've tried enabling a bunch of different messages on my ublox neo-7m in u-center but no matter what I enable it just says no data under ublox fix type/accuracy and all the height values.