ykasidit/bluetooth_gnss #25: 0 Sats in use but still get the correct position?

LeCommandeur, 2022-02-18:
Hi,

When I use the application and connect to a GNSS device via bluetooth and an NTRIP server, I do get a latitude and a longitude that pretty much correspond to my current location, but the field "N Sats used TOTAL" displays 0 and the following fields related to the different constellations display "No data" for the number of satellites in view.

This is most likely a bug or else how would I get a position that is correct?

I provided a screenshot of the application. I'm using a Samsung Galaxy S20 FE with up to date Android version if that helps.

[image]

LeCommandeur, 2022-02-21:
Hello,

Here is the NMEA Log, I hope it helps!

[2022-02-21_10-00-31_rx_log.txt](https://github.com/ykasidit/bluetooth_gnss/files/8107839/2022-02-21_10-00-31_rx_log.txt)

ykasidit, 2022-02-26:
Hello,

Just checked your nmea log and found that your device is not sending G*GSA sentences that are the source of n sats used stats. (it is only GSV so it is showing n sats in view only) If your device is a u-blox device, you can enable it in u-center.

LeCommandeur, 2022-03-03:
Yes it worked, thank you.

closed 2022-03-03 by LeCommandeur