ykasidit/bluetooth_gnss #26: Logging doesn't work (Android11) unless re-set manually

grinapo, 2022-02-27:
This may be strongly related to #16 (which seemed to be resolved).
Right now it works this way:

1. enable log directory
2. a dir selector pops up, select dir
3. accept permission from android
4. connect to BT and logging works. disconnect at the end, exit.
---
5. next time the program remembers enabled log directory and it's directory, also "propbably" remembers the permission
6. after BT connect there is _no_ logging, the display shows "no data" for all three log related status lines.
---
7. logging can be restored the following way, but have to be repeated at every start:
8. disconect (or start app)
9. switch off logging
10. switch on logging, select parent dir, select dir, accept permissions, and do as from step one.

May it be that the permission grant isn't remebered somehow?

ykasidit, 2022-10-29:
hello,

thanks for the detailed issue report

ok, please help test withl new app apk from https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.41 then go to settings > enable logging (untick and re-tick if already enabled earlier) and try again - now it is using a api call to persist the permission (issue was that the permission wont be granted from android to the same folder on next runs)

DaveInOz, 2024-08-08:
I'm currently running the latest released version of Bluetooth GNSS (1.0.49) and are experiencing the exact same issue as noted above (I am also running with an Android 11 device).  I assume the fix undertaken for v1.0.41 has migrated into the latest version (i.e. 1.0.49). If not, will there be an imminent fix?
Please advise.
Thanks

ykasidit, 2024-08-10:
Hello, thanks for the report, will try looking into it soon, was too held
up with my real work on weekdays so weekends was too lazy haha. New flutter
ver had many changes required for app to build, just staring to open today,
will try looking into this next.

ykasidit, 2024-08-13:
hello - please try new pre-release https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.51 about the logging - i tested and resolved for android 14 and 11 should work in theory too. (pls ignore the messages tab not fully ready yet)