ykasidit/bluetooth_gnss #31: GPS Garmin Montana 700i not connecting

marianalfcastro, 2022-06-27:
Hello,

I am having a persistent issue where I cannot establish a mock GPS location with my (brand new) Garmin Montana 700i. In fact, it seems I can only establish a GPS Bluetooth connection through one of their apps (Extreme, or another) and this may be the core of the issue. Considering, however, that this is not the issue and being connected through the app works, BLUETOOTH GNSS shows that all pre-connect checklist is functioning (Bluetooth powered is ON, Paired with Bluetooth Device, Target device selected, and Location is on), but then it either fails to establish a mock GPS location or it stays perpetually in the "Connecting" page. 

Any ideas on how to circumvent this? Could it be an issue with the permissions of the GPS device itself?

Many thanks!

ykasidit, 2022-06-28:
> Here is a small video showcasing the issue:
> https://youtu.be/6ch8H35iAg4
>
> Or an even shorter one
> https://imgur.com/a/HZVCFfI

Hello Fillipe,

Hope you are doing well and thanks for the issue report.

I'm not sure of this yet but let's try a few things:
- pls try go to settings of the app and try change/disable the 'secure
rfcomm connection' setting
- pls try go to settings of the app and enable 'check for settings >
location... high accuracy'
- if still fails pls try this app:
https://play.google.com/store/apps/details?id=de.pilablu.gpsconnector (some
users got that app to work when their tablet didnt work with my app - stuck
at connecting same as your video maybe just in case that might help, as I
didnt have a chance to put connection error logging into the app yet and
only 'adb logcat' collection would be required and takes time...).

ykasidit, 2022-10-29:
hello, a log would be helpful to debug the connection issue, please install new app apk from https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.41 then go to settings > enable logging (untick and re-tick if already enabled earlier) - choose a local folder to store logs, start a connection then after it fails, open your folder and zip all the contents in that log folder and share it to me to check further.