ykasidit/bluetooth_gnss #36: Wrong APK at tag v1.0.41?

IzzySoft, 2022-11-01:
The APK attached to tag v1.0.41 identifies as

    package: name='com.clearevo.bluetooth_gnss' versionCode='40' versionName='1.0.40'

Could it be you accidentally attached the wrong file (or forgot to adjust the version before building)? Would be nice if it were fixed so updates are propagated (my updater abandons the file as it already has `versionCode: 40`).

Thanks!

ykasidit, 2022-11-01:
Thanks for observing this!

Not sure how, but added a clean command now to prevent in the build_and_push_release sh, now re-released and 42.

IzzySoft, 2022-11-01:
Thanks! Looks like that worked out fine, my updater no longer complains. Speaking of it: as your app also is available [in my repo](https://apt.izzysoft.de/fdroid/index/apk/com.clearevo.bluetooth_gnss) for a while now, feel free to [grab a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) to link there from your Readme if you like :smiley:

closed 2022-11-01 by IzzySoft