ykasidit/bluetooth_gnss #38: No data reported Android 13 Galaxy S21

haliumsolutions, 2022-11-27:
App seems to work in general (mock provider) but is showing No data for time etc on the stats page:

[image]

sam4213, 2023-01-26:
Same issue with android 9 pocophone f1. Any tips to get all datas available?

ykasidit, 2023-01-29:
Hello,

Might be a parser incompatibility, please collect and share the nmea log
(enable logging in settings).

haliumsolutions, 2023-04-05:
I have an rx_log available which I am going to try to email you.

ykasidit, 2023-12-10:
Hello pls try new apkj from https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.45 and see if issue resolved or not.

haliumsolutions, 2023-12-10:
Hi,

Thanks for the update!

I downloaded this apk to test but when installing, it comes up “Bluetooth GNSS… do you want to update this app”, Hit update then Installing… short while after “App not installed as package appears to be invalid.”

?!

Kind regards,

Keith.

ykasidit, 2023-12-11:
oh ok will try test with a few more devices today

haliumsolutions, 2023-12-11:
> pls try uninstall existing app then install this new release apk from: https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.46

Ok uninstalling allowed the install to complete… Presumably deleted all the settings…

On startup it asks for permission for location and bluetooth I think but then shows message Please allow permission write_external_storage and read_external_storage…. And to reinstall if the popup doesn’t appear… I’ve tried that but am not getting a permission request for storage (presumably the configuration – as settings button is unresponsive?)

K

ykasidit, 2023-12-11:
ok noted, pls try the new apk https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.48 and confirm again.

ykasidit, 2023-12-11:
> I think we're winning :) Much less 'no data' now :> Thanks
>
> Screenshot_20231211_104320p.jpg (view on web)

Thanks for the update, great to hear that!

closed 2023-12-11 by ykasidit