ykasidit/bluetooth_gnss #62: Bluetooth permissions (android 12 and above)

kemaluzel, 2023-06-28:
It doesn't work on android 12 and above, the app closes. I guess it has to do with Bluetooth permissions.

ckuethe, 2023-10-06:
Working for me on Android 13 (pixel 6)

kemaluzel, 2023-10-06:
Have you made any changes to the permissions? I tried it on 4 different Android 12 devices, unfortunately it did not work. It's about Bluetooth permissions. I still can't figure it out

ckuethe, 2023-10-06:
[image]

kemaluzel, 2023-10-06:
https://developer.android.com/guide/topics/connectivity/bluetooth/permissions

ykasidit, 2023-12-10:
Hello, please try the new apk in the latest release (https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.45) and see if it resolves your issue or not.

kemaluzel, 2023-12-10:
Ok. I will try it on Android 12 tomorrow and let you know.

ykasidit, 2023-12-11:
Just tested on Pixel with Android 12 and yes it needs some changes, working on it and will update (the ver 45 in release wont work).

kemaluzel, 2023-12-11:
I could not install the application on the phone. Samsung Galaxy a34 5g - android 13

ykasidit, 2023-12-11:
pls try uninstall the existing bluetooth gnss app first then try again, if still fail pls show me the error screenshot

ykasidit, 2023-12-11:
Just finish adj for Android 12 - now runs successfully on Pixel 6 - pls try: https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.46

kemaluzel, 2023-12-11:
I installed v1.0.47 version on Android 13. I confirmed the permissions when the application opened. as in the screenshot below…
[image]
[image]

kemaluzel, 2023-12-11:
I installed v1.0.46 version on Android 13. Same as 47 version

ykasidit, 2023-12-11:
ok thanks, please try my fixes in https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.48 and confirm again.

kemaluzel, 2023-12-11:
I installed it and it works.
[image]
[image]
[image]

ykasidit, 2023-12-11:
> I installed it and it works.
> IMG_6951.jpeg (view on web)
> IMG_6953.jpeg (view on web)
> IMG_6955.jpeg (view on web)

Thanks for the update, great to hear this!

closed 2023-12-11 by ykasidit