ykasidit/bluetooth_gnss #64: BLE, serial data from Bluetooth LE

sigmallc, 2023-08-26:
Do you think to receive raw data via bluetooth LE and mock location.  I think you can marry your project with https://github.com/kai-morich/SimpleBluetoothLeTerminal

ser, 2024-01-26:
I think this is very very important as most of microcontrollers running micropython are unable to use Classic Bluetooth and have drivers only for BLE.

ykasidit, 2024-09-17:
hi, thanks for the suggestion. unfortunately i dont have a ble-gnss device to test yet (but i got and implemented ble qstarz racing gps) and seems there are different non-standard characteristics, i'll try do this - should not be much work as qstarz is done now - once i get a working ble gnss device.

spongebyte, 2025-03-25:
Hi. I've been using your Bluetooth GNSS app, thank you for this app!
I would like to ask if you have managed to work on a BLE implementation?

AfromD, 2025-08-11:
Thank you so much for this app. I am trying to connect a SoftRF GPS and FLARM receiver to my ereader that doesn't have GPS. I managed to jump all the hoops to get your app installed and enabled, but unfortunately, the last step fails. It won't connect to bluetooth LE. Serialbluetoothterminal works, so maybe you can get some inspiration in this example library for BLE:
https://github.com/kai-morich/SimpleBluetoothTerminal

ykasidit, 2025-08-12:
Hello,

Sorry I didn't get a device to test ble uart nmea yet, although qstarz sent
me a 10hz racing gps which is ble but it's their own protocol, not nmea.
I'll try to get time to hook it to another phone sending ble serial nmea
then, but not sure when exactly.

AfromD, 2025-08-12:
One of the cheapest external BTLE receivers is the SenseCAP Card Tracker T1000-E using SoftRF firmware.
https://www.seeedstudio.com/SenseCAP-Card-Tracker-T1000-E-for-Meshtastic-p-5913.html