ykasidit/bluetooth_gnss #71: The app is "popping upp" unintentionally as the topmost app

jjojala, 2024-05-27:
When running Bluetooth GNSS and press "overview" (or "Home") system button (e.g. to change the topmost app on the device), the Bluetooth GNSS shortly appears as the topmost app (without me, to do nothing else). This happens usually so quickly, that I've got no change to select another app in the overview view. If I'm quick enough I may be able to choose other app, but still, quickly the GNSS Bluetooth becomes topmost app again... Occasionally the other app may stay the topmost app making it's use possible. 

I've an ancient Samsung Galaxy Tab Active 2, and Honor MediaPad M5 Lite. The same problem appears on both devices. Some months ago it did not appeared with Honor (I didn't have Samsung at that time). Bluetooth GNSS version is 1.0.49. The other app, I usually want to use is OpenOrienteering Mapper.

lpechacek, 2024-05-27:
Thanks @jjojala for filing the bug report. I have the same observation and a workaround.

For completeness - steps to reproduce on my side:

1) On Samsung Galaxy Tab Active2 open the Bluetooth GNSS app
2) Start the connection
3) Press the "Recents" button (the one on the left), the list of recent apps appears
4) Select another app

Actual result: Bluetooth GNSS pops back to foreground, making the app switch impossible.
Expected result: Bluetooth GNSS stays in background. This was the behavior before.

Workaround: Press the "Home" button (the middle one), the home screen appears. Start the other app as usual by tapping its icon. Bluetooth GNSS stays in background.

jjojala, 2024-05-27:
Thanks for the worksround. It worked for me too. I do not need to take step 4 at all. App pops up before that unless I am very fast.

gaodayihao, 2024-07-26:
I also encountered the same issue.
Version 1.0.50 addresses this issue, and you can find this pre-release version in the Releases section.

ykasidit, 2024-09-17:
seems solved in v50 - thanks for the report @gaodayihao

closed 2024-09-17 by ykasidit