ykasidit/bluetooth_gnss #72: question on permissions

IzzySoft, 2024-08-18:
The scanner at IzzyOnDroid recently reported "sensitive permissions" for the app:

    android.permission.ACCESS_COARSE_LOCATION android.permission.ACCESS_FINE_LOCATION
    android.permission.BLUETOOTH_SCAN android.permission.READ_EXTERNAL_STORAGE*

The first two are clear, so I've added them to the "green list". Same for number 3:

```yaml
    android.permission.ACCESS_COARSE_LOCATION: required to access location
    android.permission.ACCESS_FINE_LOCATION: required to access location
    android.permission.BLUETOOTH_SCAN: needed to scan for Bluetooth GPS devices
```

`android.permission.READ_EXTERNAL_STORAGE*`, as indicated by the trailing asterisk, was granted implicitly as the app requests the WRITE permission. May I ask what this is needed for? With the minimum required Android version being 5, Storage Access Framework comes to mind here. Isn't that used?

And while I'm here:

    ! repo/com.clearevo.bluetooth_gnss_52.apk contains signature block blobs: 0x504b4453 (DEPENDENCY_INFO_BLOCK; GOOGLE)

That can easily be avoided with a tiny addition to your `build.gradle`:

```gradle
android {
    dependenciesInfo {
        // Disables dependency metadata when building APKs.
        includeInApk = false
        // Disables dependency metadata when building Android App Bundles.
        includeInBundle = false
    }
}
```

For some background: that BLOB is supposed to be just a binary representation of your app's dependency tree. But as it's encrypted with a public key belonging to Google, only Google can read it – and nobody else can even verify what it really contains. More details can be found e.g. here: [Ramping up security: additional APK checks are in place with the IzzyOnDroid repo](https://android.izzysoft.de/articles/named/iod-scan-apkchecks#blobs).

Thanks in advance!

IzzySoft, 2024-10-13:
@ykasidit Next release just came in and the scanner reports the same. Maybe the notification for this issue was lost?

ykasidit, 2024-10-13:
Hello,

Thanks for your recommendations. I adjusted dependenciesInfo as per your
recommendation, please check new release v58.

Regarding WRITE permission I currently understand it's required for the
'logging' feature (disabled by default) where user can enable in settings
to store the NMEA, csv of lat/lon and app operations/debug logs to a folder
in phone internal storage.

IzzySoft, 2024-10-14:
```
 iod repo get com.clearevo.bluetooth_gnss
com.clearevo.bluetooth_gnss: looking for 'https://api.github.com/repos/ykasidit/bluetooth_gnss/releases'
com.clearevo.bluetooth_gnss: checking tag 'v1.0.58'
com.clearevo.bluetooth_gnss: lastRelNo set to '1.0.58', checking for files
com.clearevo.bluetooth_gnss: Upstream file date (2024-10-14 01:21) is newer than ours (2024-10-13 19:43).
com.clearevo.bluetooth_gnss: returning ['1.0.58','https://github.com/ykasidit/bluetooth_gnss/releases/download/v1.0.58/app-release.apk',1728861694]
com.clearevo.bluetooth_gnss: 1.0.57/1.0.58, https://github.com/ykasidit/bluetooth_gnss/releases: https://github.com/ykasidit/bluetooth_gnss/releases/download/v1.0.58/app-release.apk
- Grabbing update for com.clearevo.bluetooth_gnss: OK
- Checking 'repo/com.clearevo.bluetooth_gnss_58.apk' for libraries and malware …
- Checking the app's AndroidManifest.xml …
! repo/com.clearevo.bluetooth_gnss_58.apk declares sensitive permission(s): android.permission.READ_EXTERNAL_STORAGE*
com.clearevo.bluetooth_gnss: check if repo contains FUNDING.yml
com.clearevo.bluetooth_gnss: looking for 'https://api.github.com/repos/ykasidit/bluetooth_gnss/contents/.github'
com.clearevo.bluetooth_gnss: Github reports "Not Found" for https://api.github.com/repos/ykasidit/bluetooth_gnss/contents/.github
com.clearevo.bluetooth_gnss: looking for 'https://api.github.com/repos/ykasidit/bluetooth_gnss/contents/'
com.clearevo.bluetooth_gnss: looking for 'https://api.github.com/repos/ykasidit/.github/contents/'
com.clearevo.bluetooth_gnss: Github reports "Not Found" for https://api.github.com/repos/ykasidit/.github/contents/
com.clearevo.bluetooth_gnss: no FUNDING.yml detected.
```

Blob gone, thanks! READ still there (trailing asterisk signals it was granted implicitly because of WRITE). Added both to the green list then based on your description above:

```yaml
    android.permission.ACCESS_COARSE_LOCATION: required to access location
    android.permission.ACCESS_FINE_LOCATION: required to access location
    android.permission.BLUETOOTH_SCAN: needed to scan for Bluetooth GPS devices
    android.permission.READ_EXTERNAL_STORAGE: implicitly granted because of WRITE being requested
    android.permission.WRITE_EXTERNAL_STORAGE: needed to (optionally) log NMEA data to a file of choice
```

If you want adjustments to those descriptions (or add some for others), just let me know. As my questions are now all answered (and the blog is gone), I close this issue as it's solved. Thanks a lot again!

IzzySoft, 2024-10-14:
PS: be welcome to [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) to link to [your app at IzzyOnDroid](https://apt.izzysoft.de/packages/com.clearevo.bluetooth_gnss) if you wish :wink:

closed 2024-10-14 by IzzySoft