ykasidit/bluetooth_gnss #74: External storage write permissions?

tracker51, 2024-10-03:
Hello,

So, I have successfully installed Bluetooth_GNSS on my Pixel 6a.

[image]

Likewise, I have successfully added my NTRIP connection data.

[image]

Third, I have started my RTK device (Ardusimple with ublox F9P).

Having done all this and pressing the floating button full of expectation ...

[image]

... disappointment:

[image]

What am I missing? Any help is really apptreciated!

Best regards

tracker51

ykasidit, 2024-10-13:
Hi, probably it's already resolved in a new ver - pls try new apk from: https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.57

and kindly confirm.

(I didnt update google play yet)

tracker51, 2024-10-13:
Hi, 

yes, it is! Great! Many, many thanks!

Best regards

tracker51

closed 2024-10-13 by tracker51