ykasidit/bluetooth_gnss #78: android version of the app keeps bringing itself to foreground

buratei, 2024-11-10:
I am launching this app by tasker and connecting to my GPS device via tasker intent. It works well to the point where connection is established. However, the moment my tasker sequence progresses to run the next app in line (thereby sending BT GNSS to the background), it keeps bringing itself back to the foreground with BT connection to the GPS cut. Then it automatically re-establish conblnection.

Tasker sequence ends with the BT GNSS app in the foreground, and any manual attempt to run another app is met by the same sequence described above. This behavior won't stop unless I back out to home screen, wait a few seconds before launching another app. I've tried adding delays to my tasker sequence to simulate this with no success.

buratei, 2024-11-10:
Looks like this happens even when the app is launched manually without tasker. The app doesn't play nice with the app switcher and keeps forcing itself to foreground (and subsequently tries to reconnect), unless I go to home screen wait a while. Maybe it has something to do with the default launcher that comes with the OS? The phone in question is LG V20, running latest stock Pie rom. Never had this problem on another device (nexus 7 2013).

ykasidit, 2024-11-13:
Hello,

This issue might have been resolved in the new version, please try install
the latest apk file from the release page (
https://github.com/ykasidit/bluetooth_gnss/releases/tag/v1.0.58) and let me
know if it solves the issue during app switching.

buratei, 2024-11-14:
Just checked now, it is fixed now. Job well done!
TBH, between my last posting and your update, I had discovered sending tasker intent without opening the app is a viable workaround. Still, your help is much appreciated!

ykasidit, 2025-02-06:
great! thank you for the confirmation