ykasidit/bluetooth_gnss #81: patch proposal to handle virtual http/https hosts

pbeyssac, 2025-01-25:
Currently the HTTP request generated by Bluetooh_GNSS does not include a Host: header.
This precludes the use of a caster behind a proxy or server which serves several http/https hosts.

The attached patch should probably solve this.

The patch is untested,  as I don't have a dev environment to recompile the app.

However, I have tested the rewritten request vs the current code and it fixes the issue when using the Apache proxy.

The "GET" is still labelled as "HTTP/1.0",  as announcing 1.1 instead can trigger the use of chunked transfer encoding by some casters, which is not handled by the rest of the code.

[host.txt](https://github.com/user-attachments/files/18545828/host.txt)

philippebourcier, 2025-01-26:
This is an important fix... we definitely need this.

ykasidit, 2025-01-27:
Hello,

Thanks very much for the suggestion and patch! I merged it into the code, please try the new pre-release apk v65 (just pushed) and let me know if it works for your use case.

pbeyssac, 2025-01-27:
It works like a charm! Thanks a million! Note that you can test it on my test caster, caster.ntrip.eu.org:443 (which is behind an Apache proxy).

ykasidit, 2025-02-06:
great! thank you for the confirmation

closed 2025-02-06 by ykasidit