ykasidit/bluetooth_gnss #87: Automatico GPS

Micheluzza, 2025-05-28:
Hello, could you implement an improvement?
When the application detects that it is connected to an external GPS, could it automatically connect without having to press the button? That would be very useful.

eulice124, 2025-05-31:
Nesesito conectar mi radio baufeng Uv-32

eulice124, 2025-06-02:
> Hello, could you implement an improvement?
> When the application detects that it is connected to an external GPS, could it automatically connect without having to press the button? That would be very useful.

auricgoldfinger, 2025-06-02:
If you go to settings, choose Auto connect and Autostart under Bluetooth settings. This will reconnect automatically when your device is connected to the GPS receiver