ykasidit/bluetooth_gnss #94: Points are 35 miles east of where they should be.

EcotoneCo, 2025-07-06:
Points are 35 miles east of where they should be.

ykasidit, 2025-07-09:
Hello, thanks for the issue report. Please enable logging in the app settings, connect and see the problem happen, then disconnect then please get and share me the logs (all 3 types) - you can send the logs to support@clearevo.com if you want to keep your lat/lon private.

ykasidit, 2025-07-29:
Hi, please try new release https://github.com/ykasidit/bluetooth_gnss/releases/tag/v2.0.9 and see if it helps, if it doesnt now you can long press on map to set offsets into settings according to where you press/tap on map:

https://github.com/ykasidit/bluetooth_gnss/releases/tag/v2.0.9