ykasidit/bluetooth_gnss #96: NTRIP broken since v2.0.5

Yaya-Cout, 2025-07-27:
I just upgraded to v2.0.5 (I don't know which version I had before, I didn't have the map tab before) from IzzyOnDroid.

Right after upgrade, I got an error about something being null (didn't save the error message) every time I tried to connect until I toogled "Mock location use system time" (the checkbox was just an horizontal line, like if it was in a state between on and off), then I was able to connect again to my GNSS receptor.

However, the fix type is "NORMAL" instead of the usual DGPS/FRTK/RTK, and the NTRIP card at bottom show "NTRIP Not Connected", with Server/Login filled and 0 packet received, so NTRIP/RTK is probably not working.

The full app logs are really spammy (7 line every time a location is received, with a receptor set to 200 ms), but during the connection process, I get the following error which could be related (especially the first one, modified recently https://github.com/ykasidit/bluetooth_gnss/blame/master/android/app/src/main/java/com/clearevo/libbluetooth_gnss_service/bluetooth_gnss_service.java#L233, I guess the assignation of `m_start_connect_args` got lost when it was renamed from `m_start_intent`).
I removed the MAC address from the logs

```
07-27 20:49:51.384 17288 17854 I System.out: Sun Jul 27 22:49:51 GMT+02:00 2025:DEBUG: btgnss_rfcmgr: connect() start
07-27 20:49:51.384 17288 17854 I System.out: Sun Jul 27 22:49:51 GMT+02:00 2025:DEBUG: btgnss_rfcmgr: createRfcommSocketToServiceRecord SPP_WELL_KNOWN_UUNID
07-27 20:49:51.385 17288 17854 I System.out: Sun Jul 27 22:49:51 GMT+02:00 2025:DEBUG: btgnss_rfcmgr: calling m_bluetooth_socket.connect() START m_target_bt_server_dev: name: RTK_GNSS_447 bdaddr: XX:XX:XX:XX:XX:XX
07-27 20:49:51.386 17288 17288 I System.out: Sun Jul 27 22:49:51 GMT+02:00 2025:DEBUG: btgnss_service: start_forgroud end
07-27 20:49:51.386 17288 17854 I BluetoothSocket: connect(), SocketState: INIT, mPfd: {ParcelFileDescriptor: java.io.FileDescriptor@a1cc4ba}
07-27 20:49:51.387 17288 17288 I System.out: Sun Jul 27 22:49:51 GMT+02:00 2025:DEBUG: btgnss_service: WARNING: check m_all_ntrip_params_specified exception: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.Object java.util.HashMap.get(java.lang.Object)' on a null object reference: stack_trace:
07-27 20:49:51.387 17288 17288 I System.out: 	at com.clearevo.libbluetooth_gnss_service.bluetooth_gnss_service.connect(bluetooth_gnss_service.java:233)
07-27 20:49:51.387 17288 17288 I System.out: 	at com.clearevo.libbluetooth_gnss_service.bluetooth_gnss_service$1$1.run(bluetooth_gnss_service.java:349)
07-27 20:49:51.387 17288 17288 I System.out: 	at android.os.Handler.handleCallback(Handler.java:995)
07-27 20:49:51.387 17288 17288 I System.out: 	at android.os.Handler.dispatchMessage(Handler.java:103)
07-27 20:49:51.387 17288 17288 I System.out: 	at android.os.Looper.loopOnce(Looper.java:248)
07-27 20:49:51.387 17288 17288 I System.out: 	at android.os.Looper.loop(Looper.java:338)
07-27 20:49:51.387 17288 17288 I System.out: 	at android.app.ActivityThread.main(ActivityThread.java:9106)
07-27 20:49:51.387 17288 17288 I System.out: 	at java.lang.reflect.Method.invoke(Native Method)
07-27 20:49:51.387 17288 17288 I System.out: 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:593)
07-27 20:49:51.387 17288 17288 I System.out: 	at com.android.internal.os.ExecInit.main(ExecInit.java:50)
07-27 20:49:51.387 17288 17288 I System.out: 	at com.android.internal.os.RuntimeInit.nativeFinishInit(Native Method)
07-27 20:49:51.387 17288 17288 I System.out: 	at com.android.internal.os.RuntimeInit.main(RuntimeInit.java:371)
07-27 20:49:51.387 17288 17288 I System.out: Sun Jul 27 22:49:51 GMT+02:00 2025:DEBUG: btgnss_service: m_all_ntrip_params_specified: false
07-27 20:49:52.351 17288 17854 I BluetoothSocket: connect(), socket connected. mPort=1
07-27 20:49:52.355 17288 17854 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_rfcmgr: calling m_bluetooth_socket.connect() DONE m_target_bt_server_dev: name: RTK_GNSS_447 bdaddr: XX:XX:XX:XX:XX:XX
07-27 20:49:52.356 17288 17854 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: on_rfcomm_connected()
07-27 20:49:52.358 17288 17859 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_qtowt: queue_to_outputstream_writer_thread thread start
07-27 20:49:52.358 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_istqrt: thread start
07-27 20:49:52.358 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_istqrt: readline_mode: true
07-27 20:49:52.358 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_istqrt: read_buff_mode: false
07-27 20:49:52.360 17288 17288 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: toast msg: Connected...
07-27 20:49:52.548 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: setMock gnss_ts: 1753649389000 vs system_ts: 1753649392548 = system_to_gnss_ts_diff: 3548
07-27 20:49:52.548 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location 0
07-27 20:49:52.549 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location() provider: network START
07-27 20:49:52.555 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location() provider: network SUCCESS
07-27 20:49:52.555 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location() provider: fused START
07-27 20:49:52.560 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location() provider: fused SUCCESS
07-27 20:49:52.560 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location() provider: gps START
07-27 20:49:52.562 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location() provider: gps SUCCESS
07-27 20:49:52.562 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location 2
07-27 20:49:52.562 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: activate_mock_location done
07-27 20:49:52.562 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: setmock to provider: network START
07-27 20:49:52.563 17288 17288 E flutter : [ERROR:flutter/runtime/dart_vm_initializer.cc(40)] Unhandled Exception: The location service on the device is disabled.
07-27 20:49:52.563 17288 17288 E flutter : #0      GeolocatorAndroid.getPositionStream.<anonymous closure> (package:geolocator_android/src/geolocator_android.dart:201)
07-27 20:49:52.563 17288 17288 E flutter : #1      Stream.handleError.<anonymous closure> (dart:async/stream.dart:959)
07-27 20:49:52.563 17288 17288 E flutter : #2      _HandleErrorStream._handleError (dart:async/stream_pipe.dart:303)
07-27 20:49:52.563 17288 17288 E flutter : #3      _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:188)
07-27 20:49:52.563 17288 17288 E flutter : #4      _RootZone.runBinaryGuarded (dart:async/zone.dart:1790)
07-27 20:49:52.563 17288 17288 E flutter : #5      _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:400)
07-27 20:49:52.563 17288 17288 E flutter : #6      _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:418)
07-27 20:49:52.563 17288 17288 E flutter : #7      _BufferingStreamSubscription._addError (dart:async/stream_impl.dart:322)
07-27 20:49:52.563 17288 17288 E flutter : #8      _ForwardingStreamSubscription._addError (dart:async/stream_pipe.dart:159)
07-27 20:49:52.563 17288 17288 E flutter : #9      _ForwardingStream._handleError (dart:async/stream_pipe.dart:119)
07-27 20:49:52.563 17288 17288 E flutter : #10     _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:188)
07-27 20:49:52.563 17288 17288 E flutter : #11     _RootZone.runBinaryGuarded (dart:async/zone.dart:1790)
07-27 20:49:52.563 17288 17288 E flutter : #12     _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:400)
07-27 20:49:52.563 17288 17288 E flutter : #13     _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:418)
07-27 20:49:52.563 17288 17288 E flutter : #14     _BufferingStreamSubscription._addError (dart:async/stream_impl.dart:322)
[... cut ...]
07-27 20:49:52.627 17288 17858 I System.out: Sun Jul 27 22:49:52 GMT+02:00 2025:DEBUG: btgnss_service: setMock gnss_ts: 1753649389000 vs system_ts: 1753649392627 = system_to_gnss_ts_diff: 3627
```

Thanks you for this app anyway, it saves me so much time when mapping (especially when taking pictures for Panoramax, where the phone GPS receiver have a strange form of dead reckoning which make it continue straight for some time when I turn, forcing me to manually correct the picture locations afterward, while my real GNSS receptor really work).

ykasidit, 2025-07-28:
Thanks for your time and effort in this detailed observation! I tried a fix in new 2.0.6 but dont have a login with me to try now, please try and let me know: https://github.com/ykasidit/bluetooth_gnss/releases/tag/v2.0.6

ykasidit, 2025-07-29:
Oh i didnt know centipede user/pass was public - thanks! I tried to handle the concurrentmodification error now and seems to work, please try below release and let me know:
https://github.com/ykasidit/bluetooth_gnss/releases/tag/v2.0.9

Yaya-Cout, 2025-07-29:
Yes, tested on v2.1.0 seems fine, fix type is FRTK as usual. Packet count is fixed and the margin on bottom of the screen allows to see it without having to scroll to get the data between the tabs and floating button.
The check for High Accuracy location is failing (I'm on GrapheneOS, without Google Play Services), it work with the bypass location check setting but this isn't related to this issue, I guess.

Thanks for the quick fix!

closed 2025-07-29 by Yaya-Cout